emlid/Navio2 #9: Suggestion for Navio2 C++ library

chutsu, 2016-05-09:
I've been toying around with the C++ library, and wasn't satisfied with the way the code is layed out in the repo. As a programmer I was expecting a Makefile for the Navio C++ portion that builds the library and installs it to the right place so I can link it. I instead made my own, check it out [here](https://github.com/chutsu/navio2-cpp).

It essentially builds the library and examples.

staroselskii, 2016-05-09:
@chutsu 

It'd be very nice to see a PR! We can discuss it in details there, alright?

chutsu, 2016-05-09:
Well the repo I made only contains your C++ library code, not the python and scripts. So I'm not sure a PR here makes sense.

closed 2017-02-20 by chutsu