emlid/Navio2 #12: HOW to compile C++ drivers

tizianoC, 2016-08-02:
I have tried to compile Ublox.cpp in the C++/Navio folder, running 

gcc Ublox.cpp

but it doesn't work.

May I use another compiler?
Or any other suggestions?

Thanks a lot.

staroselskii, 2016-08-02:
If you want to compile an example, ```cd``` into an Examples/<example> and ```make```
BTW, gcc is by a C compiler by default.

tizianoC, 2016-08-02:
I compiled the example but in any case I get

> Error: can not open SPI device.

The problem seems to be that I have to compile the drivers that, I think, are in the folder C ++ / Navio. I'm right?

staroselskii, 2016-08-02:
Please, follow our [docs](https://docs.emlid.com/navio2/Navio-dev/navio-repository-cloning/). Are you using our [image](https://docs.emlid.com/navio2/Navio-APM/configuring-raspberry-pi/)?

closed 2017-02-17 by staroselskii