emlid/Navio2 #14: Python GPS Example Intermittent

mr337, 2017-02-07:
I am trying the Python GPS example and getting weird results.

When testing with the C++ version I get a solid output, Long, Lat, Fix status on a timely interval (~ 1s). This verifies the GPS unit is working and getting GNSS signal resolving to a location.

But the Python GPS example just sits, every now and then will print out a solution. Sometimes does nothing for up to 10 minutes. 

Is there something wrong with the Python GPS example? I am running HEAD on Navio2 repo.

smirnov1993, 2017-02-08:
@mr337, have you tried to reboot your Navio? 
Python example can stuck if something else is trying to get access to GPS.  For example if both C++ and Python examples launched at the same time.

mr337, 2017-02-08:
@smirnov1993 Yeah and I have had that, especially with running two C++ example the second will get a UBLOX test fail, which is great!

After a clean restart via power cycle I'm still getting the same thing. Here is a console output that shows the C++ version working, nothing is running, and the python example.

```
pi@navio ~/Dev/Navio2/C++/Examples/GPS (gps_output*) $ ./gps 
SOLU TOW 320956
SOLU Longitude -93.368036
SOLU Latitude 37.194966
SOLU Height_Ellipsoid 349.662
SOLU Height_Sea 378.663
SOLU Horizontal_Accuracy 1.354
SOLU Vertical_Accuracy 2.488
GNSS Fix 1
GNSS Status 3DFIX
SOLU TOW 320957
SOLU Longitude -93.368036
SOLU Latitude 37.194966
SOLU Height_Ellipsoid 349.744
SOLU Height_Sea 378.745
SOLU Horizontal_Accuracy 1.352
SOLU Vertical_Accuracy 2.487
SOLU TOW 320958
SOLU Longitude -93.368037
SOLU Latitude 37.194965
SOLU Height_Ellipsoid 349.803
SOLU Height_Sea 378.803
SOLU Horizontal_Accuracy 1.355
SOLU Vertical_Accuracy 2.492
^C
pi@navio ~/Dev/Navio2/C++/Examples/GPS (gps_output*) $ ps aux | grep gps
pi        1203  0.0  0.2   4276  1888 pts/0    S+   17:09   0:00 grep --color=auto --exclude-dir=.bzr --exclude-dir=CVS --exclude-dir=.git --exclude-dir=.hg --exclude-dir=.svn gps
pi@navio ~/Dev/Navio2/C++/Examples/GPS (gps_output*) $ cd ../../../Python
pi@navio ~/Dev/Navio2/Python (gps_output*) $ python GPS.py 
^CTraceback (most recent call last):
  File "GPS.py", line 177, in <module>
    buffer = ubl.bus.xfer2([100])
KeyboardInterrupt
pi@navio ~/Dev/Navio2/Python (gps_output*) $ 
```

smirnov1993, 2017-02-09:
Could you please run python example with `strace`? 
Run: `strace python GPS.py`

It will give us info about which system calls failed

mr337, 2017-02-10:
Gah, so when running strace it seems to work :D 

I may have to do a screencap to show what I'm experiencing.

mr337, 2017-02-10:
So in short, when I do a `python GPS.py` it just hangs, and every now and then it will print a message.

Here is a screencap where I redacted the gps coords but you can check the timing. Literally it works when I run it with strace or attach strace to the proc. Its werid!

https://youtu.be/yZUqjoUIe9s

staroselskii, 2017-06-19:
Fixed with #17. Thanks for pointing out!

closed 2017-06-19 by staroselskii