emlid/Navio2 #24: GPS frequency heavily influenced by SPI baudrate

gmsanchez, 2017-11-06:
Hi,

I am using a Raspberry Pi 3 with a Navio2 and I want to store data from GPS and IMU at 5 Hz.

I've got an example code [GPS_AGM.py](https://gist.github.com/gmsanchez/1145c42ae6a20d2867bec42cadc2e636) (the needed code to support `UBX-NAV-PVT` messages is available in pull request #20) which calculates the elapsed time between `iTow` GPS measurements and print some relevant information. With that information in mind and expecting to have a constant sampling rate, I found out that the higher the SPI baudrate, the more inconsistent and time-varying the GPS frequency is.

If I run `top` in parallel with the Python example, I've found out the following:
* The higher the SPI baudrate is set, the more CPU % the script uses. This seems to make the Python script loose some GPS messages.
* If I use 5000000 as SPI baudrate, the CPU % is around 90% and the elapsed time between GPS measurements goes from 200 ms to 1800 ms.
* If I use 20000 as SPI baudrate, the CPU % is around 4% and the elapsed time between GPS measurements stays at 200 ms (as intended).

Could this be because of the way SPI is handled in [ublox.py](https://github.com/emlid/Navio2/blob/master/Python/navio/ublox.py)? It is a lot different from the way SPI is handled in [mpu9250.py](https://github.com/emlid/Navio2/blob/master/Python/navio/mpu9250.py) and [lsm9ds1.py](https://github.com/emlid/Navio2/blob/master/Python/navio/lsm9ds1.py), where the SPI port is opened and closed in every transaction.

drobban, 2017-12-19:
is it behaving the same way in c++ example?

gmsanchez, 2018-01-08:
@drobban sorry for the late reply, I didn't have the Navio2 with me.

I downloaded a fresh copy of Navio2 repository and compiled al the C++ examples. When running the GPS example as it is and I've got around 50% CPU usage and the **GPS Millisecond Time of Week** output is not printed at a constant rate.

However, I think that the C++ examples are not configured in the same way as the provided code, since I am receiving more than one message per clock cycle. I didn't find a function similar to [configure_solution_rate](https://github.com/emlid/Navio2/blob/master/Python/navio/ublox.py#L941) in [Ublox.cpp](https://github.com/emlid/Navio2/blob/master/C%2B%2B/Navio/Common/Ublox.cpp).

I you could provide me with some directions in order to try to make the C++ code behave the same way (just receive 1 message per clock cycle) I will try as soon as possible.

Thanks in advance!

gmsanchez, 2018-01-16:
I recently got my hands on a u-Blox NEO-M8N-0-01 GPS with serial interface. To make the comparison fair, I downloaded the drivers from [pyUblox](https://github.com/tridge/pyUblox) repository (I think you use the same for the Navio2) and after some digging around while trying to configure the UART baudrate at 115200 bps, the serial GPS works flawlessly at 10 Hz.

Could you try the code I provided? What results do you obtain?

gmsanchez, 2018-01-16:
I modified the necessary C++ files in order to get the desired behaviour. You can check out the branch **issue_24** on my Navio2 fork (available [here](https://github.com/gmsanchez/Navio2/tree/issue_24)). If my changes to the code are correct, I still can't get the GPS to work at 10 Hz (even on C++). Could you give it a look? Let me know if you have further questions. Thanks in advance!

IgorAnohin, 2018-01-18:
Hi, @gmsanchez 
I think you should change rate_ms in line 30 from 200 to 100 in [GPS_AGM.py](https://gist.github.com/gmsanchez/1145c42ae6a20d2867bec42cadc2e636). After the change GPS will work at 10Hz

gmsanchez, 2018-01-18:
Hi, @IgorAnohin.

When I set the GPS to another solution rate lower than 1000 ms (100 ms or 200 ms) the measurements do not arrive at the desired rate. If I set the solution rate to 100 ms, I receive the measurements at a variable rate: most of the time at 200 ms, sometimes every 300 ms and rarely every 100 ms.

Have you tried the provided example setting the solution rate to 100 ms? Do you get measurements constantly every 100 ms?

IgorAnohin, 2018-01-18:
@gmsanchez My steps: 
I downloaded your repo [Navio2]{https://github.com/gmsanchez/Navio2/tree/issue_24}. 
Then I downloaded [GPS_AGM.py]{https://gist.github.com/gmsanchez/1145c42ae6a20d2867bec42cadc2e636/revisions] into Navio2/Python. 
After that I changed `ubl.configure_solution_rate(rate_ms=100, nav_rate=1)`.

[This]{/home/igor/Documents/GPS_out} is my output.
Can you show me your output, please?

IgorAnohin, 2018-01-18:
@gmsanchez My steps:
I downloaded your repo [Navio2]{https://github.com/gmsanchez/Navio2/tree/issue_24}.
Then I downloaded [GPS_AGM.py]{https://gist.github.com/gmsanchez/1145c42ae6a20d2867bec42cadc2e636/revisions] into Navio2/Python.
After that I changed ubl.configure_solution_rate(rate_ms=100, nav_rate=1).

[This](https://github.com/emlid/Navio2/files/1643127/GPS_out.txt) is my output.
Can you show me your output, please?

IgorAnohin, 2018-01-18:
@gmsanchez my steps:
I downloaded your repo [Navio2](https://github.com/gmsanchez/Navio2/tree/issue_24).
Then I downloaded [GPS_AGM.py](https://gist.github.com/gmsanchez/1145c42ae6a20d2867bec42cadc2e636/revisions) into Navio2/Python.
After that I changed ubl.configure_solution_rate(rate_ms=100, nav_rate=1).

[This](https://github.com/emlid/Navio2/files/1643130/GPS_out.txt) is my output.
Can you show me your output, please?

gmsanchez, 2018-01-18:
@IgorAnohin, I followed the steps in order to set up a fresh repository. You can see my output [here](https://gist.github.com/gmsanchez/12f1fa0101ea4f36563e173e871aea60).

As you can see, the **iTow** increments are not of 100 ms (and not even constant). Could you try re running the example having 3D-fix on your GPS? Because when I got no fix, I usually get the messages every 100 ms.

Thanks in advance!

IgorAnohin, 2018-01-30:
Hi, @gmsanchez 
Thanks a lot for your patience. GLONASS & GPS can't work together on 10HZ. So i disabled GLONASS
Can you add this in [GPS_AGM.py](https://gist.github.com/gmsanchez/1145c42ae6a20d2867bec42cadc2e636)?

   

    #GLONASS
    payload = struct.pack('<BBBBBBBBI', 0, 32,32,1,6,6,12,0,0)
    ubl.send_message(0x06,0x3E, payload)
    #GPS
    payload = struct.pack('<BBBBBBBBI', 0, 32,32,1,0,8,16,0,16842753)
    ubl.send_message(0x06,0x3E, payload)

IgorAnohin, 2018-01-30:
Hi, @gmsanchez 
Thanks a lot for your patience. GLONASS & GPS can't work together on 10HZ. So i disabled GLONASS and other GNSS
Can you add this in [GPS_AGM.py](https://gist.github.com/gmsanchez/1145c42ae6a20d2867bec42cadc2e636)?
 ```
GNSS ={ ( 0, 8, 16, 0, 16842753), #GPS
            (1, 1, 3, 0, 0),          #SBAS
            (2, 2, 5, 0, 0),          #Galileo
            (3, 6, 12, 0, 0),         #BeiDou
            (4, 0, 8, 0, 0),          #IMES
            (5, 0, 3, 0, 0),          #QZSS
            (6, 6, 12, 0, 0)}         #GLONASS
 for gnss in GNSS:
        payload = struct.pack('<BBBBBBBBI', 0, 32, 32, 1, gnss[0], gnss[1], gnss[2], gnss[3], gnss[4])
        ubl.send_message(0x06,0x3E, payload)

```

staroselskii, 2018-02-26:
@gmsanchez 

Feel free to reopen if need be.

closed 2018-02-26 by staroselskii