emlid/Navio2 #30: make ADC_Navio2::initialize() return bool

chrstphrchvz, 2018-01-01:
I imagine it would be useful for [ADC_Navio2::initialize()](https://github.com/emlid/Navio2/blob/35f62ac642af921ef5f936155730ef74b423d09c/C%2B%2B/Navio/Navio2/ADC_Navio2.cpp#L21) to report to the caller whether it initializes successfully or not (similar to what [PWM:init()](https://github.com/emlid/Navio2/blob/bbea7d17b089632b60a8b2c84d9f46718cd8b952/C%2B%2B/Navio/Navio2/PWM.cpp#L8) does). Would doing so break any existing code that uses it?

staroselskii, 2018-01-02:
@chrstphrchvz 

Sounds right. Would you like to make a PR?

chrstphrchvz, 2018-11-12:
I do not currently have the means to implement or test this. If someone else would like to try, that is fine.