emlid/Navio2 #40: Information on Python version

angussmitchell, 2019-02-07:
Hi, If possible I'd like someone to add some documentation on on version of python that works with the Navio2. For example, can I run everything in Python 2 and 3? If so what version is a minimum requirement for each (i.e. is 3.5 a minimum requirement?)

staroselskii, 2019-02-07:
Hi!

If I remember correctly, it's compliant with both versions. The examples don't use any fancy Python 3 features, so I don't think that 3.5 is a must. This is usually tested on Raspbian's current version to work with both Python 2 and 3. Thanks for reminding that we need to add a .travis.yml file to the repo to actually test it :)

angussmitchell, 2019-02-07:
Thanks for the clarification! 

A related query is - what version of python (2 and 3) does the emlid raspberry image ship with? I'd love to use some of the more advanced python3 features (such as asyncio) with the Navio drivers without breaking anything.

staroselskii, 2019-02-08:
@angussmitchell 

It's Python 2.7.13 and Python 3.5.3.

angussmitchell, 2019-02-10:
Thanks for that!

angussmitchell, 2019-02-11:
Hi I'd like to reopen this issue: the Navio2 python code does not work with Python3. the obvious "print" statements don't translate to Python3. There also seems to be an issue with the smbus module. I'll post any findings I have here - I believe it won't be too hard to convert the code to Python3. 

The reason I am so concerned with using Python3, is that it is now faster than Python2 - and has other substantial advantages including concurrency using asyncio.

staroselskii, 2019-02-11:
Sure, feel free to open a PR with the changes. We'll be more than ready to review and merge them.

staroselskii, 2019-02-11:
https://github.com/emlid/Navio2/pull/27

There's also this PR with a lot of Python code than can be useful.

angussmitchell, 2019-02-20:
Thanks for the help, I'm not sure how to do a "PR" at this stage but will investigate

closed 2019-02-20 by angussmitchell