emlid/Navio2 #50: Wrong coordinate system and alignement for the python lsm9ds1 driver

hubble14567, 2020-12-22:
After implementing the madgwick algorithm on it, I noticed that the accelerometer, the gyros and the magnetometer were not lining up. The resulting coordinate system wasn't even direct.

The following code corrects the results of the accelerometer and gyro : 

```
            acc,gyr,mag=(self.imu2.getMotion9())
            acc[0],acc[1]=-acc[1],-acc[0]
            gyr[2]=-gyr[2]
            gyr[0], gyr[1] = gyr[1], gyr[0]
```

Now, for the acc. and the gyro. : the z axis points toward the ground, x follows the arrow on the PCB and y is such that the coordinate system is orthogonal direct.

I don't know how to check if the magnetometer is aligned, but after calibrating it (and correcting the acc and gyro), the madgwick algorithm gave me the right pitch, roll and yaw, with very good performances

AlexanderDranitsa, 2021-08-03:
Hi @hubble14567 
Thank you! 
We'll take a look at this soon. Is this all changes you propose to add? May I suggest using a PR for this?

hubble14567, 2021-08-03:
Hi,
Is this all changes you propose to add? Yes
May I suggest using a PR for this? If you want, yes you can do so.

I never (realy) used github before, this comment was a while ago and I don't have my Navio anymore (so I cannot troubleshoot and create a good PR myself). I simply posted this to quickly help other people with a problem I solved. Do whatever your think is best with my original post.

AlexanderDranitsa, 2021-08-03:
Got it, thanks.

gmsanchez, 2021-08-03:
@AlexanderDranitsa @hubble14567 the LSM9DS1 driver outputs is data to match the MPU9250 axis orientation. Could be that the source of the issue? This behavior kind of confusing and it is not well documented (that is why the axis change may seem quite arbitrary), but I think it is made this way so the AHRS algorithms supplied by Emlid work with any of the two IMU.

hubble14567, 2021-08-03:
@gmsanchez I don't remember using the MPU9250 (had some issues, and LSM9DS1 was working fine), so I cannot tell you if the axis of both match. Also did not use the AHRS from Navio. Your theory is possible.

Nonetheless the coordinate system is highly confusing, and I would also change it on the MPU9250 if it is the same. The norm for coordinate systems should be x forward (that's the arrow one the Navio), z down and (most importantly) y in a way that creates an orthogonal direct coordinate system. LSM9DS1 coordinates system for the acc and gyro were a mess in that regard and were not even the same.

gmsanchez, 2021-08-03:
Something similar happened to me a while ago, when I needed the axis to conform to ENU. I suggest you download both the LSM9DS1 and the MPU9250 datasheet and see how their axis are oriented on the Navio2. Then try to apply the LSM9DS1 axis the same operations as the ones that are made in the driver. After that, they should match the MPU9250 or at least you can make sure where each axis is pointing.